LanguidSmartass/mdb-arduino-cashless #9: How to charge money on card?

pistoletov1974, 2016-04-12:
Hi i just try on same priject - vending with rfid. I suppouse (IMHO)  that using socket server have more advantages for this project.  How users put money on cards?

LanguidSmartass, 2016-04-12:
Hello.
SIM900 has built-in HTTP handling functions, so it was much easier to read server responses.
I only needed the HTTP Code and a string representing available user funds.
Yes, you can also use standard TCP/IP socket connection and make a request, but you're gonna have to parse the entire TCP response, which includes a lot of headers in HTML syntax, such as <tr>...</tr>, and it requires a bigger buffer ==> less available SRAM. SIM900 can do all of this by itself.
The idea initially was not to store funds on RFID cards, but on the remote server (that's another task to complete). The card is just an ID holder.
I had some thoughts of implementing account refilling via coin acceptor and/or bill validator of the VMC itself, but haven't done it yet. Didn't find anything similar in MDB 4.2 Specification either.

pistoletov1974, 2016-04-13:
I think a lot about using rfid and where store funds on server or on card.Server is good idea - but if we have problems with connection? Another way is to use NFC cards. What do you think about it? We can store funds on card. I think to put change on card on the vmc. IMHO pure socket connection is lightweight than http - this is pure our data without any headers. We need to write our own socket server but this is really simple using java sockets. GPS trackers use pure socket connections. Http produce lot unnecessary trafic.

LanguidSmartass, 2016-04-14:
If we really have connection problems -- than do nothing. Having funds on the card can lead to a lot of issues described in this article http://cybergibbons.com/uncategorized/cashless-vending-fail-analysis-of-a-rfid-payment-system/
You didn't get it about http -- you don't have to implement anything except for simple AT commands wrapper when working with SIM900. I use Arduino (Atmega328P) and GPRS Shield (SIM900). I didn't write any socket.c/h libraries, all this functionality is built-in in SIM900. I'm only sending and receiving bytes via SoftwareSerial, that's all, SIM900 does all the job.
Though if you use Ethernet shield (Wiznet W5100 chip) or some WiFi interface then yes -- sockets are the right way.

LanguidSmartass, 2016-05-26:
Issue closed as more than 1 month passed after the last post.

closed 2016-05-26 by LanguidSmartass

Kirson75, 2020-09-19:
Hi,
I am using part of your code and it works successfully on an old coffee machine. G-snack in the machine (it is also MDB protocol)  all blocks the keyboard on the coffee machine.

LanguidSmartass, 2020-09-21:
Can you please rephrase the last sentence? I don't quite get what you're trying to say.

Kirson75, 2020-09-22:
Thank

I overcame a lot of problems, and on one coffee shop he still works full.

But G-snack Coffe machine blocked. Arduino's  blocks even the keyboard to
the device.

 I Find  reason:
String:
file MDB.c
case VMC_RESET: / * PORTC ^ = (1 << 0); * / MDB_ResetHandler (); перерыв;

Frequencies: Reset ();

What can be the problem?

mkrasselt1, 2020-09-28:
arduino blocking sounds like an loop going cracy, did you change anything on the hardware ? some transmission wires bad ? or did you change the software ? 
i skiped using this project and implemented everthing new - but you error sounds like something i had in the early stages... length counter = -1 => Sending Overflow and stuff like that.
Start by Outputing all the recieved and send data to the console to spot the mistakes

LanguidSmartass, 2020-09-28:
Dude, this project is pretty much abandoned and serves as an example of a
simplest state machine for level 1 mdb cashless devices.

It was debugged on a single MDB master available to me at the time, thus
any "standard compliance" issues are to be expected.

I don't have neither the energy nor the desire to continue developing it.

However, you can always branch/fork it and continue the work that is left
behind if you so desire.