LanguidSmartass/mdb-arduino-cashless #10: Use on Arduino Mega

mkrasselt1, 2017-07-04:
Dear Languid,
I am Trying to use your software on an Arduino Mega.
Where do i have to change the Pin Numbers for Serial1?
Thanks a Lot, your Software is very usefull

LanguidSmartass, 2017-07-04:
Hello.
To port my code onto ATmega2560 you need to reconfigure USART1 into 9-bit
mode (1 start, 9 data bits, and whatever parity/stop bits in compliance
with the MDB protocol). Configuring 9-bit data frame via arduino libraries
is impossible, you will have to access ATmega2560 registers directly,
google the datasheet and look into the Universal Syncronous/Asyncronous
Receiver/Transmitter 0-3, your C++-like object called Serial1 is mapped to
the TX1 and RX1 pins on the MCU, which corresponds to USART1 peripheral
module.
So you basically need to do some changes in USART.c and USART_ISR.c

mkrasselt1, 2017-07-10:
So thanks for your Help, now i now what and where to change the code, but i am having a little issue trying to get it to work even on an arduino uno. Maily because USART_ISR.c is not included in any(or where do i have to put it) and even for arduino uni it seems wrong ? (I had to change lines to 
`ISR(USART0_UDRE_vect)`
to get it compiled.
Do you have an Idea whats wrong with it ? Debug output only tells me:
`Start up
Ethernet ON
0
`

LanguidSmartass, 2017-07-27:
Sorry for the delay.
USART_ISR.c can't be included anywhere, because it's not a header file. It is automatically picked by the compiler, if you provided the directory in paths list.
There is not enough info for me to operate, try to look into the standard and throw some frames into your VMC. Also check the hardware connection, which should contain optocouplers.

closed 2019-03-23 by LanguidSmartass