LanguidSmartass/mdb-arduino-cashless #15: File Location

aksD3V, 2018-09-06:
Dear Languid,
Where do I have to paste/replace all the ".c" and ".h" files. 
I am using Arduino IDE 1.8.6 and the only location where I am able to locate USART.c and USART.h is at the folder location 

C:\Program Files (x86)\Arduino\hardware\arduino\avr\firmwares\wifishield\wifi_dnld\src\SOFTWARE_FRAMEWORK\DRIVERS\USART

Please help me with the folder locations for the files.

Also I am trying to port this code to Arduino Mega. Please let me know which file I have to edit for 9 bit serial mode.

LanguidSmartass, 2019-09-19:
See this function in a reference file, the 9th bit is set there:
void USART_Init(uint16_t baud_rate)
[https://github.com/LanguidSmartass/mdb-arduino-cashless/blob/master/USART.c](url)

As for the files you're talking about -- they're seem to be from a different driver for some wifi shield.
USART.c and USART.h that are used in my MDB implementation are located in the repository.

eyupnarinc, 2019-09-21:
Hi Languid
I get about 50 errors as follows. what is the problem? what should I do? please help. Arduino IDE version 1.8.9. I tried arduino uno and mega 
Thank you from now

C:\Users\EYP~1\AppData\Local\Temp\arduino_build_21082\libraries\mdb-arduino-cashless-master\MDB.c.o (symbol from plugin): In function `vmc_config':

(.text+0x0): multiple definition of `vmc_config'

C:\Users\EYP~1\AppData\Local\Temp\arduino_build_21082\sketch\MDB.c.o (symbol from plugin):(.text+0x0): first defined here

C:\Users\EYP~1\AppData\Local\Temp\arduino_build_21082\libraries\mdb-arduino-cashless-master\MDB.c.o (symbol from plugin): In function `vmc_config':

(.text+0x0): multiple definition of `vmc_prices'

C:\Users\EYP~1\AppData\Local\Temp\arduino_build_21082\sketch\MDB.c.o (symbol from plugin):(.text+0x0): first defined here

C:\Users\EYP~1\AppData\Local\Temp\arduino_build_21082\libraries\mdb-arduino-cashless-master\MDB.c.o (symbol from plugin): In function `vmc_config':

(.text+0x0): multiple definition of `csh_error_code'

C:\Users\EYP~1\AppData\Local\Temp\arduino_build_21082\sketch\MDB.c.o (symbol from plugin):(.text+0x0): first defined here

C:\Users\EYP~1\AppData\Local\Temp\arduino_build_21082\libraries\mdb-arduino-cashless-master\MDB.c.o (symbol from plugin): In function `vmc_config':

(.text+0x0): multiple definition of `csh_state'

C:\Users\EYP~1\AppData\Local\Temp\arduino_build_21082\sketch\MDB.c.o (symbol from plugin):(.text+0x0): first defined here

C:\Users\EYP~1\AppData\Local\Temp\arduino_build_21082\libraries\mdb-arduino-cashless-master\MDB.c.o (symbol from plugin): In function `vmc_config':

(.text+0x0): multiple definition of `csh_poll_state'

C:\Users\EYP~1\AppData\Local\Temp\arduino_build_21082\sketch\MDB.c.o (symbol from plugin):(.text+0x0): first defined here

C:\Users\EYP~1\AppData\Local\Temp\arduino_build_21082\libraries\mdb-arduino-cashless-master\MDB.c.o (symbol from plugin): In function `vmc_config':

(.text+0x0): multiple definition of `csh_config'

C:\Users\EYP~1\AppData\Local\Temp\arduino_build_21082\sketch\MDB.c.o (symbol from plugin):(.text+0x0): first defined here

C:\Users\EYP~1\AppData\Local\Temp\arduino_build_21082\libraries\mdb-arduino-cashless-master\MDB.c.o (symbol from plugin): In function `vmc_config':

(.text+0x0): multiple definition of `vend_amount'

C:\Users\EYP~1\AppData\Local\Temp\arduino_build_21082\sketch\MDB.c.o (symbol from plugin):(.text+0x0): first defined here

C:\Users\EYP~1\AppData\Local\Temp\arduino_build_21082\libraries\mdb-arduino-cashless-master\MDB.c.o (symbol from plugin): In function `vmc_config':

(.text+0x0): multiple definition of `user_funds'

C:\Users\EYP~1\AppData\Local\Temp\arduino_build_21082\sketch\MDB.c.o (symbol from plugin):(.text+0x0): first defined here

C:\Users\EYP~1\AppData\Local\Temp\arduino_build_21082\libraries\mdb-arduino-cashless-master\MDB.c.o (symbol from plugin): In function `vmc_config':

(.text+0x0): multiple definition of `item_cost'

C:\Users\EYP~1\AppData\Local\Temp\arduino_build_21082\sketch\MDB.c.o (symbol from plugin):(.text+0x0): first defined here

C:\Users\EYP~1\AppData\Local\Temp\arduino_build_21082\libraries\mdb-arduino-cashless-master\MDB.c.o (symbol from plugin): In function `vmc_config':

(.text+0x0): multiple definition of `MDB_Init'

C:\Users\EYP~1\AppData\Local\Temp\arduino_build_21082\sketch\MDB.c.o (symbol from plugin):(.text+0x0): first defined here

C:\Users\EYP~1\AppData\Local\Temp\arduino_build_21082\libraries\mdb-arduino-cashless-master\MDB.c.o (symbol from plugin): In function `vmc_config':

(.text+0x0): multiple definition of `MDB_Send'

C:\Users\EYP~1\AppData\Local\Temp\arduino_build_21082\sketch\MDB.c.o (symbol from plugin):(.text+0x0): first defined here

[... 210 lines cut ...]
C:\Users\EYP~1\AppData\Local\Temp\arduino_build_21082\libraries\mdb-arduino-cashless-master\ringBuf.c.o (symbol from plugin): In function `rBufInit':

(.text+0x0): multiple definition of `rBufIsFull'

C:\Users\EYP~1\AppData\Local\Temp\arduino_build_21082\sketch\ringBuf.c.o (symbol from plugin):(.text+0x0): first defined here

C:\Users\EYP~1\AppData\Local\Temp\arduino_build_21082\libraries\mdb-arduino-cashless-master\ringBuf.c.o (symbol from plugin): In function `rBufInit':

(.text+0x0): multiple definition of `rBufElemCount'

C:\Users\EYP~1\AppData\Local\Temp\arduino_build_21082\sketch\ringBuf.c.o (symbol from plugin):(.text+0x0): first defined here

HardwareSerial0.cpp.o (symbol from plugin): In function `Serial':

(.text+0x0): multiple definition of `__vector_18'

C:\Users\EYP~1\AppData\Local\Temp\arduino_build_21082\sketch\USART_ISR.c.o (symbol from plugin):(.text+0x0): first defined here

HardwareSerial0.cpp.o (symbol from plugin): In function `Serial':

(.text+0x0): multiple definition of `__vector_19'

C:\Users\EYP~1\AppData\Local\Temp\arduino_build_21082\sketch\USART_ISR.c.o (symbol from plugin):(.text+0x0): first defined here

collect2.exe: error: ld returned 1 exit status

LanguidSmartass, 2019-09-21:
Those a linker errors.

Somehow you've got into situation where you're trying to link together same object files (.c.o files in this case), thus the "multiple definition of _some_symbol_" pops up.

I'm not much into Arduino IDE, but I would suggest you to clean up any temporaries you can find related to this build.

LanguidSmartass, 2019-09-21:
Actually, on the second view of your log I can probably see why this is happening:

Let's take "ringBuf.c.o" for example -- that's an object file compiled from "ringBuf.c" source file.

You have two versions of it, one is at
"C:\Users\EYP~ 1\AppData\Local\Temp\arduino_build_21082\sketch\"
and the second is at
"C:\Users\EYP~ 1\AppData\Local\Temp\arduino_build_21082\libraries\mdb-arduino-cashless-master\"

Since they're essentialy the same files, when linker tries to put them together into a final binary it throws the errors I described above, because both of them contain the definitions of the identical symbols (namely, functions and variables).

Maybe try to **back up** and then delete C:\Users\EYP~1\AppData\Local\Temp\arduino_build_21082\ directory and start anew.

eyupnarinc, 2019-09-21:
First of all thank you for answering
I did but I get the same errors. what else can I do

LanguidSmartass, 2019-09-21:
Just a general advice, since I can't tell you anything more specific:

Try to figure out which files linker tries to put together and why there a multiple compilation units of the same source files in the first place.

eyupnarinc, 2019-09-21:
thank you i will try again and give information

eyupnarinc, 2019-09-21:
I cleared temp files and libraries
now gives these errors. Do you have any idea about this?
////////////////////////////////////////////////////////////////////////////////////////////////
HardwareSerial0.cpp.o (symbol from plugin): In function `Serial':

(.text+0x0): multiple definition of `__vector_18'

C:\Users\EYP~1\AppData\Local\Temp\arduino_build_763332\libraries\MDB\USART_ISR.c.o (symbol from plugin):(.text+0x0): first defined here

HardwareSerial0.cpp.o (symbol from plugin): In function `Serial':

(.text+0x0): multiple definition of `__vector_19'

C:\Users\EYP~1\AppData\Local\Temp\arduino_build_763332\libraries\MDB\USART_ISR.c.o (symbol from plugin):(.text+0x0): first defined here

collect2.exe: error: ld returned 1 exit status

LanguidSmartass, 2019-09-21:
Ok, those ones are actually related to this project.

Arduino's toolchain uses __vector_18 and __vector_19 symbols for it's own UART handlers, namely HardwareSerial class works with them deep deep down.

Because there is no 9bit frame support in HardwareSerial, I implemented my own UART init and handlers in "USART.c" and "USART_ISR.c", and the definitions of the same vectors conflict with the Arduino ones.

You need to find those in the Arduino headers/sources and comment them out, then recompile.

Also, if you need a debugging tool you can use SoftwareSerial class and a USB-UART dongle, since you won't be able to use HardwareSerial.

LanguidSmartass, 2019-09-24:
Did you resolve your issue?

eyupnarinc, 2019-09-24:
THANK YOU
I solved this problem and cleaned up the libraries.
I have 2 more problems :)
1. RFID card does not read
2. On the server side to verify the card id how a file must be. Do you have a sample

LanguidSmartass, 2019-09-24:
I think you've gotta figure out those ones yourself, since those are not exactly the issues concerning this particular project.
A couple of suggestions though: 
- check your physical SPI connection and CS pins assignment, there is an SD card and MFRC reader that both use this interface;
- server side is completely up to you to implement, I used simple GET/POST HTTP requests with some bootleg signature, a web server that processed these requests and MSSQL database to which the server made translated requests in consequence.

closed 2019-09-24 by LanguidSmartass