LanguidSmartass/mdb-arduino-cashless #17: undeclared vars

llino, 2019-05-30:
when compiling, I am getting undefined vars:
 PORTC, in MDB.c 
`switch (csh_state)
    {
        case CSH_S_INACTIVE : /*PORTC = 0; PORTC |= (1 << 0);*/ break;
        case CSH_S_DISABLED : /*PORTC = 0; PORTC |= (1 << 1);*/ break;
        case CSH_S_ENABLED  : PORTC = 0; PORTC |= (1 << 2); break;
        case CSH_S_SESSION_IDLE : PORTC = 0; PORTC |= (1 << 3); break;
        case CSH_S_VEND     : PORTC = 0; PORTC |= (1 << 4); break;
        default : break;
    }`

UCSR0A, in USUART.c

is there any setting that I am missing?

Thanks

LanguidSmartass, 2019-05-30:
That's strange.
Try to check for "#include <avr/io.h>", UCSR0A should be defined somewhere up the chain of includes.
Can you please be more specific on your toolchain?

llino, 2019-05-30:
Thanks for the quick response!
I was compiling for the wrong board... I was compiling for esp8266 instead of arduino Uno :(
In Arduino IDE changing now to Uno board.

llino, 2019-05-30:
Can confirm that changing board to Uno, MDB_SlaveAutonomous.ino compiles sucessfully.
Thanks for the help and great job with this project 💯

closed 2019-06-03 by LanguidSmartass