LanguidSmartass/mdb-arduino-cashless #19: Webserver code

harleygroupmap, 2021-05-25:
Hi
Thanks for our great job
But can you please share the code of webserver project?
Seems like it's simple one, but it's hard to rebuild it from arduino code..

mkrasselt1, 2021-05-26:
Do start a Webserver of your Choice:
Add Subfolder to Match this URL:
http://xxx.xxx.xxx.xxx:xxx/api/Vending/
A file for each Card UID contains the Credit for each card, named like the uid without any ending
http://xxx.xxx.xxx.xxx:xxx/api/Vending/A42F32DD should return 4000 as the user funds in cents
http://xxx.xxx.xxx.xxx:xxx/api/Vending/A42F32DD/100 should decrease the user funds by 100 cents

thats it - not very advanced but its a proofe of concept... for my company we advanced this stuff by *3000 and are using esp32 and a websocket connection and other fine stuff :)

LanguidSmartass, 2021-05-26:
> One day i might be able to share my/our improved code for free :))

Hello.

As you said, this code is purely a prototyp and the real device was a
single proof of concept, never going into any type of production.

You can organise the API and other client-server communication details as
your heart desires :)

That's basically the reason why there aren't any schematics, since anyone
interested can guess about these by reading MDB 4.2, pin assignments in the
code, etc. That would take a couple of days of work at most.

closed 2022-05-31 by LanguidSmartass