LanguidSmartass/mdb-arduino-cashless #21: Vmc_commands

amsfec32, 2022-02-14:
Hello awesome code written for cashless device.
I have some problems in my case. My device gets reader enable command and according to program device goes to enabled state so vmc should not issue reader enable command until reader gets disabled. But my device getting reader enable multiple times. Second is, when reader issues a beginsession command and vmc should ack, nack, ret in the response of beginsession. But in my case my device is getting 0x12 byte which is poll. And sometimes session begun and  sometimes session does not begin. Do we need to set poll status to beginsession until device does not receive ack. Same with vendapproved and endsession command. Pls help
Thankyou

amsfec32, 2022-02-15:
Thanks for you response.
When i issue beginsession command with available user funds, then sometimes vmc begins the session and show user credit on lcd but some times it ignores beginsession command so as i stydy in nama mdb spec4.2 that vmc does ack the sessionbegin command so i want to use that ack, nak, ret send by vmc to confirm that sessionbegin and if nak is received by reader then sessionbegin should be issue again.

mkrasselt1, 2022-03-12:
i am not shure if i do understand correctly but yes, if vmc does not ack your session begin, you should do it again :)
we implemented as such, that the cashless device answers every poll with session begin, as long as the vmc did not change to session idle yet

closed 2023-09-26 by LanguidSmartass