LanguidSmartass/mdb-arduino-cashless #22: Schematic

radarcontact, 2022-04-04:
Do you have any sample schematics or pictures of how you connected it up? 

(I know there is an example schematic in the MDB spec. I was hoping to see how you specifically did it with component model numbers. If you're willing to share and provide an answer it would be very helpful. Thank you in advance.)

mkrasselt1, 2022-04-04:
We started with the example code of Nottinghack hacklabs - like this one
https://wiki.nottinghack.org.uk/images/a/a2/VMC-Nanode_Schematic.png

mkrasselt1, 2022-04-04:
But 2 Regulat optocoplers with matching resistors should work as well

mkrasselt1, 2022-04-05:
But 2 Regular optocoplers with matching resistors should work as well

malekses, 2022-05-26:
If you can share the schematic diagram you will be appropriated.
I really need to do exactly same as your project.
Thanks in advance.

malekses, 2022-05-29:
> We started with the example code of Nottinghack hacklabs - like this one https://wiki.nottinghack.org.uk/images/a/a2/VMC-Nanode_Schematic.png

Hi, thank you for your reply, could you please share another schematic with more clarifications? some of the items like capacitors does not mentioned the value of it, I'm new in Arduino and I need to do this immediately.
thanks.

mkrasselt1, 2022-05-29:
So first of all, you need nothing and especially not immediatly from us - we are here to help you get startet and not provide a ready to use solution for you. Do you agree on that?
The capacitor you mentioned is only part of the power supply - you could skip that for now - if you power your arduino using 5v USB power.
Otherwise - place the lm2596 board tuned to 5v or the fixed output variant in the typical application example circuit from the datasheet there.
If you dont understand what i am talking about, you should start with this one here:
https://www.instructables.com/Make-a-DeviceFeature-for-Vending-Machines/

malekses, 2022-06-02:
> So first of all, you need nothing and especially not immediatly from us - we are here to help you get startet and not provide a ready to use solution for you. Do you agree on that? The capacitor you [...]

I know that you are here to help me get started and not provide a ready to use solution my friend, but the Arduino with MDB interface is very hard to find over the internet, I told you above I'm new in Arduino, and I need some support to build a graduation project, and it's not a big issue if you share a small schematic diagram with us even handwritten if you cannot; could you please explain why the Ethernet Shield W5100 needed in this project?

LanguidSmartass, 2022-06-02:
1) This code I wrote years ago is supposed to be an example to be torn
apart, I didn't guarantee it to work out of the box.

2) Hence, the Ethernet shield is not required to run it – these calls to
w5100 hal are just to demonstrate that "here's a place for your user code".

3) I'd recommend you ditch the Arduino software environment entirely and
use C/C++ (avr-gcc with either Atmel/Microchip studio or CLion and a
configured CMake toolchain), maybe even a custom build of avr-gcc 11 (it
does exist, but not on the microchip's website).

4) The principal optocoupling schematic is in the standard – I didn't
provide any since I assumed that anyone who stumbles upon this obscure
repository will RTFM :)
In short, you physically connect your MCU's UART Tx to an IRED input of one
optocoupler through a resistor, and Rx – to an optotransistor's collector
of the second optocoupler and pull it up. That's it, the simple version of
the schematic contains only 2 optocouplers and 3 resistors. But again –
RTFM.

5) Grab the usart/mdb files, extract the state machine from "Slave.ino",
compare them to the MDB protocol document (link is in the README),
understand how the communication works and embed all this into your own
code.

P.S.:
Since this repo got a good lot of attention over the years, considering how
niche the protocol itself is, I do have an intent of providing a complete
rework. A portable Embedded C++ library – with an example schematic, maybe
a simple PCB layout.

However, I can't say when I'm gonna do it, since I don't have access to a
real VMC anymore to test the code. Unless I implement a VMC master as well
just for lolz.

malekses, 2022-06-02:
> STM32

thanks for reply,
If you interested to help, I will pay you.
please contact me here malek.safadi@gmail.com

mkrasselt1, 2022-06-03:
@LanguidSmartass we are able too indeed - we are using the rmt feature to do so - it works great on any pin - because it makes use of a seperate co prozessor designd fpr data transmission and reception :))

LanguidSmartass, 2022-06-03:
@mkrasselt1 Oh, I actually didn't know about that feature on esp – thanks

mkrasselt1, 2022-08-10:
https://github.com/mkrasselt1/esp32-mdb-master

Took some time but here it is, take a look at the lib/mdb/driver part of it ;)

mkrasselt1, 2022-08-10:
and to finaly close this issue - on a 3.3V arduino you can use this schematic:
https://github.com/mkrasselt1/esp32-mdb-master/blob/master/pcb/MDB-Slave-Interface.PNG

closed 2022-11-06 by LanguidSmartass