LanguidSmartass/mdb-arduino-cashless #23: hello, i have a question

marevaloz, 2022-11-19:
hi, i am very surprised by your work, congratulations... i have a doubt, when you connect the arduino to the vmc using the interface, you connect using the serial port (arduino usb), rx0/tx0, or the debug pins that you define using the softwareserial library? I see that you overwrote the UART port for 9bits but since there is no picture of how you did it, I am not sure where you send the signal to the vmc from. Thanks!

LanguidSmartass, 2022-11-19:
Yeah, if you're utilizing an Uno or any other board that has only one
hardware uart, there's no other way than to ditch the Serial class
completely and use your own initialization. However, if you have a Mega
with 3 free hw uarts, you can leave the uart0 for usb debugging purposes

marevaloz, 2022-11-19:
OK thank you! mmmm...my doubt is due to the fact that I saw output messages to softwareserial , "texts" sent to this port and I thought "maybe I should use this port as a debugging monitor and rx0/rx0 to join it to the mdb interface", for example This line

Debug.print(F("Funds: "));
     Debug.println(user_funds);
    
     Debug.println(F("Poll State: Begin Session"));
     CSH_SetUserFunds(user_funds);
     CSH_SetPollState(CSH_BEGIN_SESSION);

then, the library that manages the communication with the vmc (mdb.c) uses the other port (not softwareserial), and, in the debug pins I can debug...or should I comment out the debug lines...because I go through of this port

Thanks for helping me understand your code better!

marevaloz, 2022-11-19:
now, remove the call to the serial library as you indicate, but in the case of the library for the mfrc522 reader, it tells me that it cannot send outputs by serial, should they still be commented or do you use a particular library?

I use this, version 1.4.9

https://github.com/miguelbalboa/rfid

and this is the error

C:\Users\user\Documents\Arduino\libraries\MFRC522\src\MFRC522.cpp: In member function 'void MFRC522::PCD_DumpVersionToSerial()':
C:\Users\user\Documents\Arduino\libraries\MFRC522\src\MFRC522.cpp:1352:2: error: 'Serial' was not declared in this scope
  Serial.print(F("Firmware Version: 0x"));
  ^~~~~~
C:\Users\user\Documents\Arduino\libraries\MFRC522\src\MFRC522.cpp:1352:2: note: suggested alternative: 'String'
  Serial.print(F("Firmware Version: 0x"));
  ^~~~~~
  String
C:\Users\user\Documents\Arduino\libraries\MFRC522\src\MFRC522.cpp:1353:18: error: 'HEX' was not declared in this scope
  Serial.print(v, HEX);
                  ^~~
C:\Users\user\Documents\Arduino\libraries\MFRC522\src\MFRC522Extended.cpp: In member function 'MFRC522::StatusCode MFRC522Extended::TCL_Transceive(MFRC522Extended::PcbBlock*, MFRC522Extended::PcbBlock*)':
C:\Users\user\Documents\Arduino\libraries\MFRC522\src\MFRC522Extended.cpp:729:3: error: 'Serial' was not declared in this scope
   Serial.print("CRC is not taken care of by MFRC522: ");
   ^~~~~~
C:\Users\user\Documents\Arduino\libraries\MFRC522\src\MFRC522.cpp: In member function 'void MFRC522::PICC_DumpToSerial(MFRC522::Uid*)':
C:\Users\user\Documents\Arduino\libraries\MFRC522\src\MFRC522.cpp:1402:4: error: 'Serial' was not declared in this scope
    Serial.println(F("Dumping memory contents not implemented for that PICC type."));
    ^~~~~~
C:\Users\user\Documents\Arduino\libraries\MFRC522\src\MFRC522.cpp:1402:4: note: suggested alternative: 'String'
    Serial.println(F("Dumping memory contents not implemented for that PICC type."));
    ^~~~~~
    String
C:\Users\user\Documents\Arduino\libraries\MFRC522\src\MFRC522Extended.cpp:729:3: note: suggested alternative: 'String'
   Serial.print("CRC is not taken care of by MFRC522: ");
   ^~~~~~
   String
C:\Users\user\Documents\Arduino\libraries\MFRC522\src\MFRC522.cpp:1411:2: error: 'Serial' was not declared in this scope
  Serial.println();
  ^~~~~~
C:\Users\user\Documents\Arduino\libraries\MFRC522\src\MFRC522Extended.cpp:730:29: error: 'HEX' was not declared in this scope
   Serial.println(rxModeReg, HEX);
                             ^~~
C:\Users\user\Documents\Arduino\libraries\MFRC522\src\MFRC522.cpp:1411:2: note: suggested alternative: 'String'
  Serial.println();
  ^~~~~~
  String
C:\Users\user\Documents\Arduino\libraries\MFRC522\src\MFRC522.cpp: In member function 'void MFRC522::PICC_DumpDetailsToSerial(MFRC522::Uid*)':
C:\Users\user\Documents\Arduino\libraries\MFRC522\src\MFRC522.cpp:1421:2: error: 'Serial' was not declared in this scope
  Serial.print(F("Card UID:"));
  ^~~~~~
C:\Users\user\Documents\Arduino\libraries\MFRC522\src\MFRC522.cpp:1421:2: note: suggested alternative: 'String'
  Serial.print(F("Card UID:"));
  ^~~~~~
  String
C:\Users\user\Documents\Arduino\libraries\MFRC522\src\MFRC522.cpp:1427:33: error: 'HEX' was not declared in this scope
   Serial.print(uid->uidByte[i], HEX);
                                 ^~~
C:\Users\user\Documents\Arduino\libraries\MFRC522\src\MFRC522Extended.cpp: In member function 'void MFRC522Extended::PICC_DumpToSerial(MFRC522Extended::TagInfo*)':
C:\Users\user\Documents\Arduino\libraries\MFRC522\src\MFRC522Extended.cpp:1015:4: error: 'Serial' was not declared in this scope
    Serial.println(F("Dumping memory contents not implemented for that PICC type."));
    ^~~~~~
C:\Users\user\Documents\Arduino\libraries\MFRC522\src\MFRC522.cpp:1435:27: error: 'HEX' was not declared in this scope
  Serial.println(uid->sak, HEX);
                           ^~~
C:\Users\user\Documents\Arduino\libraries\MFRC522\src\MFRC522.cpp: In member function 'void MFRC522::PICC_DumpMifareClassicToSerial(MFRC522::Uid*, MFRC522::PICC_Type, MFRC522::MIFARE_Key*)':
C:\Users\user\Documents\Arduino\libraries\MFRC522\src\MFRC522.cpp:1474:3: error: 'Serial' was not declared in this scope
   Serial.println(F("Sector Block   0  1  2  3   4  5  6  7   8  9 10 11  12 13 14 15  AccessBits"));
   ^~~~~~
C:\Users\user\Documents\Arduino\libraries\MFRC522\src\MFRC522Extended.cpp:1015:4: note: suggested alternative: 'String'
    Serial.println(F("Dumping memory contents not implemented for that PICC type."));
    ^~~~~~
[... 183 lines cut ...]
    Serial.println(F("Activating the UID backdoor failed."));
    ^~~~~~
C:\Users\user\Documents\Arduino\libraries\MFRC522\src\MFRC522.cpp:1851:4: note: suggested alternative: 'String'
    Serial.println(F("Activating the UID backdoor failed."));
    ^~~~~~
    String
C:\Users\user\Documents\Arduino\libraries\MFRC522\src\MFRC522.cpp:1860:4: error: 'Serial' was not declared in this scope
    Serial.print(F("MIFARE_Write() failed: "));
    ^~~~~~
C:\Users\user\Documents\Arduino\libraries\MFRC522\src\MFRC522.cpp:1860:4: note: suggested alternative: 'String'
    Serial.print(F("MIFARE_Write() failed: "));
    ^~~~~~
    String
C:\Users\user\Documents\Arduino\libraries\MFRC522\src\MFRC522.cpp: In member function 'bool MFRC522::MIFARE_UnbrickUidSector(bool)':
C:\Users\user\Documents\Arduino\libraries\MFRC522\src\MFRC522.cpp:1886:4: error: 'Serial' was not declared in this scope
    Serial.print(F("MIFARE_Write() failed: "));
    ^~~~~~
C:\Users\user\Documents\Arduino\libraries\MFRC522\src\MFRC522.cpp:1886:4: note: suggested alternative: 'String'
    Serial.print(F("MIFARE_Write() failed: "));
    ^~~~~~
    String

exit status 1

Compilation error: exit status 1

LanguidSmartass, 2022-11-20:
I don't know much about the intrinsics of stock arduino mfrc library, but
maybe it has an option to disable serial output by some macro definition.

If not, you'd probably be better off with a custom mfrc lib that doesn't
use Serial, or just go and comment it's usage out in the arduino library
itself.

marevaloz, 2022-12-19:
Hello!!, so long!, check, the connection of the cashless device, did you connect it directly to the communication port of the MDB or did you implement some electronic interface between the arduino device, do you have a diagram of how it should be connected correctly?
Do you have a draft of how their connection should be? I use a wafer cable (http://www.waferstar.com/en/MDB2PC-PC2MDB.html), which I bought for my tests and it works very well with the biller and coiner, but these devices are connected via mdb cable powered at 24 volts, in the case of the arduino I don't use the power cables but I do use the communication cables, I think I just have to understand how it should be connected so that I can control it through the 10H or 60H port, can you help me with this?

LanguidSmartass, 2022-12-19:
Hi.

I've built a simple schematic with two optocouplers (for rx/tx lines), 2 or
3 current limiting resistors and a DC-DC converter module to switch from 24
to 5V. The connection was done as it is described in the standard, I don't
remember the page but it's somewhere at the end of the protocol section.

marevaloz, 2022-12-19:
hey so fast! how cool you are! Did you invert the signals with the rx/tx octacouplers? what octacoupler did you use? I am using the 4n35 that can be connected in an inverted or direct way, personally I have used it directly

LanguidSmartass, 2022-12-20:
I think I just used the old toshiba pc817 ones in DIP4 package, the whole
thing was assembled on a through -hole breadboard.

Don't remember much about inversion though – again, the schematic in the
standard will help you more than I right now.

amsfec32, 2023-01-12:
> 

Hello marevaloz,
i hope you are doing well. You will need to invert the signal wire which is directed towards slave device.

closed 2023-09-26 by LanguidSmartass