LanguidSmartass/mdb-arduino-cashless #24: Command disorder

amsfec32, 2023-01-12:
Hello, 
I am using an arduino as a cashless device but i have a problem of command disorder, when arduino do reply of poll by a session begin response then suddenly vmc issues reset command and when reader comes to enable state then a proper vend occurs.. pls help

mkrasselt1, 2023-01-12:
could you provide a log of your last messages to the vmc ? did you got enabled first ?

LanguidSmartass, 2023-01-12:
Hi, have you checked against the command sequences in the standard? How
does your case differ exactly, can you put up a diagram yourself for the
sake of clarity?

amsfec32, 2023-01-13:
The stansard sequesnce is reset, poll, setup config, setup max/min, exp req id and then reader enable in my case everything was correct but when reader was enable poll ack was going then my reader do reply of a poll with the session begin responce then vmc issues reset command but accounding to data sheet vmc should issue vend request command.

mkrasselt1, 2023-09-26:
no activity here, will close it.
@amsfec32 might reopen it, when he writes again

closed 2024-05-14 by mkrasselt1