LanguidSmartass/mdb-arduino-cashless #26: Wiring Arduino Uno to MDB Connector Molex

tomeroto, 2023-09-13:
Hi friend, I have Arduino Uno and MDB connector from Vending Machine. How to connected it, do you have schema / wiring of diagram to make your code works?

mkrasselt1, 2023-09-25:
just take a look at the mdb specsheet - there is an example with two optocouplers. connect them to the rx and tx pins of the slave device

LanguidSmartass, 2023-09-26:
Made a pull-request with a kicad schematic made in about an hour.

@mkrasselt1 I've sent you an invitation to be a contributor (because github won't let me merge the branch myself).
And since you were more active on this repository through the years than myself, it would probably be better anyway.

mkrasselt1, 2023-09-26:
@LanguidSmartass 
Yes thank you, i might work on this a little from time to time.
But @tomeroto  https://github.com/LanguidSmartass/mdb-arduino-cashless/blob/master/mdb-interface.kicad_sch
this is the file he is talking about - open in kicad latest version :)

closed 2023-09-26 by mkrasselt1