LanguidSmartass/mdb-arduino-cashless #28: vmc never asks for min max price setup, keeps repeating setup step

DragonHunter274, 2024-05-18:
The VMC never is stuck in a loop of poll and setup, I've tried a few different settings in setup already, but as the vmc ACKs it I'm not sure whats going on.
Screenshot of logic analyzer attached, I'm  unable to get a proper export unfortunately

[image]
I hope someone can help me with this, I'm stuck

Tchoum, 2024-05-18:
Hello,

The 162 / 62 is the poll for the cashless device 2. 
It seems like the VMC doesn't care about your response, or your response is incorrectly configured.

The response should be within 5 ms, but I think it works because you received the config command.
Can you confirm if the reset command works fine?

Maybe the response is incorrect. You configured the country code for the euro. I think the scale factor should be something like 0x01 (for 1 cent €). 
The MISC options are set to 9 (1001).
Test with the MISC options set to 0x01 or 0b00000001 and another test with the feature level set to 0x02.

DragonHunter274, 2024-05-18:
I tried with the misc options set to 0x01 and scale factor 0x01, haven't tried feature level 0x02 yet,
This is my reset:
[image]

I can upload the whole capture but it can only be opened by Digilent Waveforms and it is a pretty big file

DragonHunter274, 2024-05-18:
Here is the whole thing including installers for Waveforms for all major platforms and the code I'm currently running: https://hub.dh274.com/s/ToDrppzJjPWWaXL

Tchoum, 2024-05-18:
it appears you aren't reading the checksum byte from the reset. I think you need to read h110 and h010, and only after that, respond with an ACK (h100).
I can't install waveforms for now, but lets try this:
in the "void Reset(void)" function, before the line "MDB_Send(CSH_ACK);"
add this : 

uint16_t readCmd;
while (1)
        if (MDB_DataCount() > 0)
            break;
        MDB_Read(&readCmd);

Tchoum, 2024-05-18:
With your WaveForms log, I noticed something:
you have an extra delay when responding to command h111, around ~19 ms, while the manual says the maximum response time is 5 ms.
Can you comment out line 75 (delay(15)) and see if it changes anything?

Tchoum, 2024-05-18:
With your WaveForms log, I noticed something:
you have an extra delay when responding to command h111, around 19 ms~, while the manual says the maximum response time is 5 ms.
Move, after all the MDB_send command, the Debug.println("reader config") line 780. is consuming time.  At 9600 baud, 14*1.145ms = 16ms~.
increase speed of debug.begin if possible.
or disable debug with #define SOFTWARESERIAL_DEBUG_ENABLED 0

DragonHunter274, 2024-05-20:
adding ` uint16_t readCmd; while (1) if (MDB_DataCount() > 0) break; MDB_Read(&readCmd);`doesnt change anything, it still sends just reset at the same time as before

mkrasselt1, 2024-05-20:
Could you please provide your complete code somehow?

DragonHunter274, 2024-05-20:
I just uploaded the current code to https://hub.dh274.com/s/ToDrppzJjPWWaXL (same link as above)

mkrasselt1, 2024-05-22:
i noticed in the beginning you send feature level 1 - some machines wont support this, you better bump that to feature level two in setup config data (0x11) and it might help to increase Z7 - maximum response time - to maximum or a suitable middle value for now, to make shure its not the source

tomeroto, 2024-05-23:
hi friend  can you share schematic that you use in this work. Thanks before.

DragonHunter274, 2024-05-23:
[image]
I just noticed it responds to poll with CSH_ACK but I think it should respond with CSH_JUST_RESET

Tchoum, 2024-05-23:
Hi, I see a little conflict in the Reset() function and the JustReset() function in response to a VMC reset command or VMC poll command.

Try this:
- in JustReset() uncomment line 772 (`uint16_t readCmd; while (1) if (MDB_DataCount() > 0) break; MDB_Read(&readCmd);`)
- in Reset() comment/delete line 763 `uint16_t readCmd; while (1) if (MDB_DataCount() > 0) break; MDB_Read(&readCmd);.`
- add this line in MDB_ResetHandler(void) befor Reset(); line 538.

try that and update the file log in your nextcloud storage.

and yes, like mkrasselt1say, try featureLevel 2,  but i think you schould wirte 0x02.

DragonHunter274, 2024-05-23:
Now it doesn't respond to reset at all

I will upload my code to github I think thats easier